naturerobots/move_base_flex #187: Implement the getRobotVelocity function

q576333, 2020-04-23:
Hi everyone
Now, I try to implement my own local planner, so I need the actual robot velocity feedback. Therefore I try to contribute getRobotVelocity function in RobotInformation class. But I see the highlight in this function comment is "TODO implement and filter tf data to compute velocity.". What's the meaning in this sentence: "filter tf data to compute velocity"? Does it mean using the amount of change in two consecutive pose(from map to base_link) to find the robot speed?
Why not direct subscribe the /odom topic to get velocity feedback?

q576333, 2020-04-24:
I have already prepared PR.
In addition, I have confused why we need two getRobotPose function? One in the RobotInformation class, another in the mbf_utility namespace. I think getRobotVelocity function encounters the same situation.

corot, 2020-04-24:
I think yes, that's what the TODO means. But Subscribing to odom is a better option, for sure. I would do this way.

As for getRobotPose. the one in RobotInformation is a wrapper for the one in mbf_utility. Probalby you should use the one in RobotInformation, and implement the method to get the velocity from odom topic. (actually there's A THIRD version on AbstractNavigationServer but I'll remove it).

@spuetz , thoughts?