naturerobots/move_base_flex #190: Move mbf_msgs into it's own git repo

jspricke, 2020-04-27:
Hi,

currently mbf_msgs are part of the mbf repo, meaning that they get a new release whenever you do a new mbf release. That's not needed and is irritation cause messages are an external interface.
Please move them to a new repo.

spuetz, 2020-04-27:
Hey ;) Are there more reasons for doing that? Because, we have a couple of other issues to solve and I like it to have it all in the same repo. I wouldn't count the versioning as a strong argument. :D Cheers.

jspricke, 2020-04-27:
hm.. breaks our internal tooling ;)

Rayman, 2020-05-01:
For the maintainers it saves a lot of time keeping everything in one repo

jspricke, 2020-05-01:
@Rayman that's not really true, also look at other ROS packages, like common_msgs, control_msgs or moveit_msgs.