naturerobots/move_base_flex #192: action constant variable cannot be found

hoppss, 2020-05-09:
move_base_flex/mbf_abstract_nav/src/recovery_action.cpp: In member function ‘void mbf_abstract_nav::RecoveryAction::run(mbf_abstract_nav::AbstractActionBase<mbf_msgs::RecoveryAction_<std::allocator<void> >, mbf_abstract_nav::AbstractRecoveryExecution>::GoalHandle&, mbf_abstract_nav::AbstractRecoveryExecution&)’:

move_base_flex/mbf_abstract_nav/src/recovery_action.cpp:71:52: error: ‘STOPPED’ is not a member of ‘mbf_msgs::RecoveryResult {aka mbf_msgs::RecoveryResult_<std::allocator<void> >}’
         result.outcome = mbf_msgs::RecoveryResult::STOPPED;

move_base_flex/mbf_abstract_nav/src/controller_action.cpp: In member function ‘void mbf_abstract_nav::ControllerAction::run(mbf_abstract_nav::AbstractActionBase<mbf_msgs::ExePathAction_<std::allocator<void> >, mbf_abstract_nav::AbstractControllerExecution>::GoalHandle&, mbf_abstract_nav::AbstractControllerExecution&)’:

move_base_flex/mbf_abstract_nav/src/controller_action.cpp:189:51: error: ‘STOPPED’ is not a member of ‘mbf_msgs::ExePathResult {aka mbf_msgs::ExePathResult_<std::allocator<void> >}’
         result.outcome = mbf_msgs::ExePathResult::STOPPED;
_____
these constance variable defined in .action cannot be found, move_base_flex can compile successfully in my pc ,but failed in my tx2

spuetz, 2020-06-12:
What compiler are you using at your tx2?

hoppss, 2020-06-16:
yes

hoppss, 2020-06-24:
sorry , this one tx2 had been reset, now I find it is ok to compile for ubuntu18 && gcc 7.5 && tx2

closed 2020-06-24 by hoppss