naturerobots/move_base_flex #193: Release move_base_flex into noetic?

croesmann, 2020-05-12:
Hey guys!

Is move_base_flex ready to be released into noetic?
It's currently, the only blocking package for the teb_local_planner.

Cheers, Chris

spuetz, 2020-05-12:
Hey Chris, nice to hear from you. I can to that in the next days or on the weekend. Cheers ;)

doisyg, 2020-05-25:
That would be nice to release it now noetic is out. Mbf compiles without issue

spuetz, 2020-05-25:
Hi, I'm on it! Sorry, for the delay.

croesmann, 2020-05-25:
no problem ;-)
I just saw that the travis build for noetic is failing. I guess you need to switch from the legacy branch in [industrial-ci](https://github.com/ros-industrial/industrial_ci) to the master branch in your .travis.yml by checking it out explicitly.

closed 2020-05-26 by spuetz

spuetz, 2020-08-04:
See #216