naturerobots/move_base_flex #202: Logging verbosity

reinzor, 2020-06-30:
Would you be open for a pull request that reduces the logging? We are in a situation that we plan and update paths often; this results in really verbose logging:

```
Action "exe_path" canceled
Start action "exe_path" using controller
..
..
```

Examples:

- https://github.com/magazino/move_base_flex/blob/73155af186988644b5d334cf6cfdb2434257f929/mbf_abstract_nav/src/abstract_navigation_server.cpp#L213
- https://github.com/magazino/move_base_flex/blob/73155af186988644b5d334cf6cfdb2434257f929/mbf_abstract_nav/src/abstract_navigation_server.cpp#L237

Would it be possible to change these lines to `DEBUG`? Or should we introduce a parameter `action_event_log_level` or something?

Rayman, 2020-07-01:
Just an idea, you can also change logger level of nodes with a configuration file, this is explained here: http://wiki.ros.org/rosconsole#Configuration

For example:
Create the file `$ROS_ROOT/config/rosconsole.config` with the following content:
```
# Set the default ros output to warning and higher
log4j.logger.ros=WARN
# Override my package to output everything
log4j.logger.ros.my_package_name=DEBUG
```

reinzor, 2020-07-01:
Oh, nice. Didn't know this. Thanks for the info!

corot, 2020-07-06:
I'm still in favor or changing to DEBUG `"Start action ...` msgs, cause with continuous replanning are very spamming;
but possibly keep the cancel as INFO, as they are more meaningful events.

reinzor, 2020-07-06:
@corot , thanks for your response? Also the controller cancel? This will be triggered every time the simple action server receives a new goal. For the planner I can image this is desired:

- Move base goal: info
- Move base cancel: info
- Recovery goal: debug
- Recovery cancel: debug
- Controller goal: debug
- Controller cancel: debug
- Planner goal: debug
- Planner cancel: info

Could you update this according to your preference?

corot, 2020-07-06:
:thinking:  but each time you send a new path to a running controller, there's no cancel;  just set the new path (as long as you use the same concurrency_slot and the same controller).
Is not the case? Please double check that you are not sending a cancel inadvertently.

reinzor, 2020-07-06:
Hmm, we indeed send a cancel request (automatically sent by the simple actionclient).I will update the PR.

corot, 2020-07-06:
> Hmm, we indeed send a cancel request (automatically sent by the simple actionclient).I will update the PR.

But that will introduce a small discontinuity in the velocity commands... even if you set `force_stop_on_cancel` to false.
Better to do as I say: update ongoing exe_path action

closed 2020-07-06 by reinzor