naturerobots/move_base_flex #212: mbf doesn't cancel when returning BLOCKED_PATH in computeVelocityCommands

Timple, 2020-07-16:
move_base_flex doesn't cancel the action when returning BLOCKED_PATH in computeVelocityCommands.

I would expect this to happen [here](https://github.com/magazino/move_base_flex/blob/f64a77317043cd42d7449522edb2fd2a342df01c/mbf_abstract_nav/src/abstract_controller_execution.cpp#L381) somewhere. But instead only [NO_LOCAL_CMD](https://github.com/magazino/move_base_flex/blob/f64a77317043cd42d7449522edb2fd2a342df01c/mbf_abstract_nav/src/abstract_controller_execution.cpp#L397) remains of all the error options that can be given.

I would expect a cancel when an error in the controller occurs. Am I missing something or is this functionality missing?

corot, 2020-07-16:
Are you calling the move_base action?
If you are instead calling the exe_path action, it should fail with
a BLOCKED_PATH outcome.

One thing that could be happening is that controller_max_retries is not 0,
and so MBF retries calling computeVelocityCmd and it succeeds after some
retries.

Timple, 2020-07-16:
> controller_max_retries is not 0

Ah, this was one part of the problem.

The other was that the `cancel()` function is called even though we just send a failure and no goals where active. We didn't expect this and thus also didn't handle this very nice. That's solved now! 

Thanks for your assistance, was our bug after all.

closed 2020-07-16 by Timple