naturerobots/move_base_flex #216: A new Noetic release with API changes planned

spuetz, 2020-08-04:
Who is using Noetic in production? We discussed the release and API issues in a maintainer meeting with Magazino. If no one, or only one or two are using noetic jet in production, we are planning to integrate API chnages into the next noetic release. See #193 and #191.

doisyg, 2020-08-04:
We do. Thanks for the notification.

spuetz, 2020-08-04:
@doisyg Would it help to collect the API chnages here? So you could preapre the changes at you site.

doisyg, 2020-08-04:
Definitly!

Rayman, 2020-08-04:
We don't use noetic yet

Timple, 2020-08-04:
We don't yet. Migration is in the works but API changes don't hurt us.

reinzor, 2020-08-19:
> Who is using Noetic in production?

Thanks for the notification. We are not using Noetic in production yet.

spuetz, 2020-09-16:
The release will be a bit delayed.