naturerobots/move_base_flex #224: move_base_legacy_relay.py throws error on Jetson Nano (ROS Melodic)

jash101, 2020-10-10:
Related to issue #222, but I thought it's better to open a separate issue for this.
I tried @Rayman's turtlebot_mbf implementation (https://github.com/Rayman/turtlebot3_mbf) on Kinetic and it worked fine. But on Melodic it throws this error:
```
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/mbf_costmap_nav/move_base_legacy_relay.py", line 129, in <module>
    mb_drs = Server(MoveBaseConfig, mb_reconf_cb)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/dynamic_reconfigure/server.py", line 81, in __init__
    self._change_config(self.config, ~0)  # Consistent with the C++ API, the callback gets called with level=~0 (i.e. -1)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/dynamic_reconfigure/server.py", line 104, in _change_config
    self.config = self.callback(config, level)
  File "/opt/ros/melodic/lib/mbf_costmap_nav/move_base_legacy_relay.py", line 101, in mb_reconf_cb
    mbf_drc.update_configuration(mbf_config)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/dynamic_reconfigure/client.py", line 185, in update_configuration
    raise DynamicReconfigureParameterException('don\'t know parameter: %s' % name)
dynamic_reconfigure.DynamicReconfigureParameterException: don't know parameter: make_plan_add_unreachable_goal
```

Due to this, I'm unable to run move_base_flex on Melodic. Has someone else faced the same issue on Melodic? Or is it Jetson Nano specific?

darshb34, 2020-10-14:
yes @jash101 I am also facing the same issue with `    <node pkg="mbf_costmap_nav" type="move_base_legacy_relay.py" name="move_base_legacy_relay"/>
`

corot, 2020-10-14:
Yes, also happens in melodic. It's due to a couple of new parameters added to the move_base dynamic configuration.
Was fixed on Aug 4th, but not yet released. We will do soon, but meanwhile, you can use source code from noetic-devel or melodic-devel. Or even path your melodic installation by replacing the script at /opt/ros/melodic/lib/mbf_costmap_nav/ and deleting the .pyc version.

jash101, 2020-10-14:
Hello @corot I tried melodic, noetic and noetic-devel. Only noetic-devel seems to have the issue fixed, the other two branches give the same error. (I couldn't find melodic-devel). But when I ran @Rayman's launch file from https://github.com/Rayman/turtlebot3_mbf (which works fine in Kinetic), I get following error:
`[FATAL] [1602699058.992457608, 199.320000000]: The states RECALLED and REJECTED are not implemented in the SimpleActionServer!`
Any idea how to solve it?

spuetz, 2020-10-14:
The code on `master` should also contain the fix. I think this was the branch @corot meant with melodic-devel. We'll release a new version soon.

corot, 2020-10-15:
> The code on `master` should also contain the fix. I think this was the branch @corot meant with melodic-devel. We'll release a new version soon.

Yes, sorry; I meant master

corot, 2020-10-15:
> The states RECALLED and REJECTED are not implemented in the SimpleActionServer!

The move_base goal is either RECALLED or REJECTED. No idea about that...  this happens in Noetic? melodic? both?

darshb34, 2020-10-15:
Thank you so much for fast responses @corot @spuetz .
I tried copy-pasting the legacy code from the repo to the path you mentioned replacing the original file with proper permissions. 

I am on ROS-MELODIC. The code in both 'master' and 'noetic-devel' branch gives me `The states RECALLED and REJECTED are not implemented in the SimpleActionServer!` error.

corot, 2020-10-15:
How you send goals?  Rviz 2D nav goal?

jash101, 2020-10-15:
@corot yes, I'm sending goals via Rviz. I'm using ROS Melodic and have built from source

darshb34, 2020-10-22:
@jash101 @corot is the error `"The states RECALLED and REJECTED are not implemented in the SimpleActionServer!" ` related to move_base_flex or something else specific to Jetson?

corot, 2020-10-22:
I can only tell that is the first time I see either RECALLED or REJECTED
states....
so could be an issue with the Jetson (never used one neither, sorry)

darshb34, 2020-10-23:
Thanks for the response @corot  

 I figured that in `simple_goal_cb()` the below send_goal action, 

`mbf_mb_ac.send_goal(mbf_msgs.MoveBaseGoal(target_pose=msg, controller=blp, planner=bgp))`

results into the below error,

` WARN] [1603443831.271168744]: No plugin loaded with the given name "global_planner/GlobalPlanner"!
[FATAL] [1603443831.271950381]: The states RECALLED and REJECTED are not implemented in the SimpleActionServer!
`
But if I replace it with `mbf_mb_ac.send_goal(mbf_msgs.MoveBaseGoal(target_pose=msg)` , I do not get any errors and things work fine.

corot, 2020-10-26:
I see; REJECTED makes sense in that case, as MBF cannot execute the goal, and so rejects it.
I'll provide better result for that cases (#227)

jash101, 2020-10-31:
I think implementing mbf the way it is done in celeres navigation by @spuetz is the way to go, at least in Jetson Nano

corot, 2020-11-06:
Current release should fix this issue. Closing

closed 2020-11-06 by corot