naturerobots/move_base_flex #226: get_path-Action returns INVALID_PLUGIN-errorcode

ML-S, 2020-10-14:
I'm trying to call the get_path-Action inside a SMACH. When calling the action, it returns the **error code 59**, which is according to the get_path-Actionserver the code for '**INVALID_PLUGIN**'.
My **mbf-launchfile** looks as follows:
```
<?xml version="1.0"?>

<launch>
  <!-- Move Base Flex -->
  <master auto="start"/>
  <node pkg="rosservice" type="rosservice" name="set_move_base_log_level" args="call --wait /move_base_flex/set_logger_level 'ros.mbf_abstract_nav.actionlib' 'debug'" />
  <node pkg="mbf_costmap_nav" type="mbf_costmap_nav" respawn="false" name="move_base_flex" output="screen">
    <param name="tf_timeout" value="1.5"/>
    <param name="planner_max_retries" value="3"/>
    <rosparam file="$(find joggingblinde_2dnav)/config/base_global_planner_params.yaml" command="load" />
    <rosparam file="$(find joggingblinde_2dnav)/config/base_local_planner_params.yaml" command="load" />
    <rosparam file="$(find joggingblinde_2dnav)/config/costmap_common_params.yaml" command="load" ns="global_costmap"/>
    <rosparam file="$(find joggingblinde_2dnav)/config/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find joggingblinde_2dnav)/config/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find joggingblinde_2dnav)/config/local_costmap_params.yaml" command="load" />
  </node>

</launch> 
```
The **costmap_commmon_params.yaml** looks as follows:
```
footprint: [[0.20, 0.24125], [0.20, 0.175], [0.65, 0.175], [0.65, -0.175], [0.20, -0.175], [0.20, -0.24125], [-0.20, -0.24125], [-0.20, -0.175], [-0.65, -0.175], [-0.65, 0.175], [-0.20, 0.175], [-0.20, 0.24125]] # jogging person at regular speed
# robot_radius: 0.5
inflation_radius: 0.1
# cost_scaling_factor:
transform_tolerance: 0.05
```
The **global_costmap_params.yaml** looks as follows:
```
global_costmap:
  plugins: 
    - {name: static_layer, type: "costmap_2d::StaticLayer"}
    - {name: inflation_layer, type: "costmap_2d::InflationLayer"}
  track_unknown_space: true
  global_frame: map #according to ros base_local_planner page it should be the same frame as the odometry runs in (/odom atm).
  robot_base_frame: base_link
  update_frequency: 10.0 # data comes in
  publish_frequency: 10.0 # costmap publishes info
  rolling_window: true
  recovery_behavior: false
  static_layer:
      trinary_costmap: false
      map_topic: "path_segmentation_occgrid"
```
And the **base_global_planner_params.yaml** looks as follows:
```
GlobalPlanner:
  #basic parameters
  default_tolerance: 3.0  #default is 0.0
  use_dijkstra: true  #default is true
  old_navfn_behavior: false  #default is false
  use_quadratic: true  #default is true
  use_grid_path: false  #default is false
  allow_unknown: true  #default is true
  lethal_cost: 253  #default is 253
  neutral_cost: 50  #default is 50
  cost_factor: 3  #default is 3

  #debug parameters
  publish_potential: True
```

All plugins worked with regular move_base. All plugins were loaded successfully without error when calling the mbf-launchfile.
Has somebody any idea what could be the issue?

spuetz, 2020-10-14:
You need to specify which plugins to load:  How to you call the `GetPath` action? For exmaple in another yaml config file
```
planners:
  - name: 'global_planner'
    type: 'global_planner/GlobalPlanner'
```

Next you can and should specify the planner in the action goal message `planner="global_planner"`. This allows you to call different loaded planners at runtime. MBF lists the loaded plugins at program start.

ML-S, 2020-10-15:
That solved it, thank you!

closed 2020-10-15 by spuetz

ML-S, 2020-10-16:
One more question:
In your [ceres_robot-project](https://github.com/uos/ceres_robot/tree/master/ceres_navigation) you specified the plugins as follows:

```
planners:
  - name: 'GlobalPlanner'
    type: 'global_planner/GlobalPlanner'
  - name: 'SBPLLatticePlanner'
    type: 'SBPLLatticePlanner'
```
Is _SBPLLatticePlanner_ for global or local planning? I'm asking because I'm trying to use _dwa_local_planner/DWAPlannerROS_ for local planning and therefore replaced the _SBPLLatticePlanner_ above with it, which results in an error that it couldn't find the DWAPlannerROS-plugin.