naturerobots/move_base_flex #230: Only compute one trajectory with exePath

ML-S, 2020-10-27:
Hello again,

I'm still trying to implement a SMACH that uses mbf for a controlled navigation on ROS Melodic. My goal is to really controll every step of the process, meaning that when I call exePath, I only want to compute exactly 1 trajectory. 

Let's say my SMACH looks as follows: 
[image]

In this example, my global map updates every 5 seconds, which means that also my global path has to update. Concurrently, I want to plan my local plan (based on the current global plan stored in a variable) with DWAPlannerROS using the ExePath-Action and send the resulting velocity-command to a feedback-state. 

Currently I'm calling the exePath-Action like this:
```
# result-callback for execPath-ActionService (Local Path)
def execPathCallback(userdata, status, result):
    if result.outcome == ExePathResult.SUCCESS:
        ...some result handling...
        return 'succeeded'
    elif result.outcome == ExePathResult.CANCELED:
        return 'preempted'
    else:
        ...some error handling...
        return 'aborted'

# callback-function to pass path to local planner
def set_execpath_goal_callback(userdata, goal):
    goal = need to do some stuff here, therefore callback
    return goal

...

smach.StateMachine.add('EXEC_PATH_STATE', 
                        smach_ros.SimpleActionState(
                            '/move_base_flex/exe_path',
                            ExePathAction,
                            result_cb=execPathCallback,
                            goal_cb=set_execpath_goal_callback,
                            input_keys = ['global_path']),
                            transitions={'succeeded': 'FEEDBACK_STATE',
                                         'aborted': 'aborted',
                                         'preempted': 'preempted'})

smach.StateMachine.add('FEEDBACK_STATE', ...)
...
```

The way I'm doing it right now, it appears that the result_cb is not called until the whole path is executed (the goal is reached) or an error occured, is that right?

My question is: is it possible to call the ExePath-Action in order to compute only one Feedback-Trajectory? If so, how would I do that? If not, what could I do as a workaround and would it be useful to implement a possibility to do so in terms of full flexibility?

corot, 2020-10-28:
> The way I'm doing it right now, it appears that the result_cb is not called until the whole path is executed (the goal is reached) or an error occured, is that right?

yes

> My question is: is it possible to call the ExePath-Action in order to compute only one Feedback-Trajectory?

no; exe_path expects a path and will execute it until done/error/cancel

> If so, how would I do that? If not, what could I do as a workaround and would it be useful to implement a possibility to do so in terms of full flexibility?

Not sure if i get what you want to implement,,, but if for you is enough to "do something" on each feedback callback while exe_path runs, easy solution is to extend smach_ros.SimpleActionState and override _goal_feedback_cb method. I do that here:
https://github.com/corot/thorp/blob/master/thorp_smach/src/thorp_smach/toolkit/navigation_states.py#L157

ML-S, 2020-10-28:
@corot , thank you for your fast reply! 

Well, the thing is that I want as less stuff as possible run simultaneously in my SMACH. The system I implement sets a goal which is like 30 meters away from the robot. Every few meters (e.g. every 5 seconds) the global map as well as the global path towards this global goal is updated. Therefore, I don't have this usual process in which I compute a global map, set a goal, compute the global path and then follow this path until the goal is reached but have a frequently updating global path and want to control every step (also meaning every iteration of the local planner) instead of having the robot automatically being navigated to the goal without being able to influence this process (by for example updating the global path or even the goal itself). 

But overriding the _goal_feedback_cb-method really could help in that case, I'll definitely try that! However, a combination of overriding the _goal_feedback_cb and cancelling the running ExePath-Action when I received the first feedback should give me exactly the behaviour I need. Is there a propery working way to reliably cancel the ExePath-Action (meaning that it cancels immediately without getting stuck)?

corot, 2020-10-29:
But calling exe_path with a new path while already running will not stop the robot. exe_path will just adopt the new path and start following it (unless you change the controller or the concurrency_slot).
Is not what you need?

ML-S, 2020-10-29:
Ah, that's good to know! Well, I mis-interpreted the exe-path-Action at first...but I just changed my approach a little in order to use exe_path the way you suggested it. 

How would I update the global path on a running exepath-Action? You mentioned calling exe_path again while it's running would do so, but how am I supposed to call it again, when it's running and therefore doesn't return until it's finished? Is there a method which I could use to change the given exepath-goal while the action is running? (What I mean is: I started exepath with a global path as its goal. Exepath is now trying to navigate the robot along this path and won't return until it's finished. In the meanwhile, the global path gets recalcualted in a concurrent state. Now I need a possibility to "update" the global path which the running exepath-Action is currently using as its goal so that the exepath will continue following the new calculated global path)

corot, 2020-10-30:
Just send a new goal. It will preempt the previous one, without stopping
the robot

ML-S, 2020-11-01:
So you're suggesting calling the `sendGoal`-function like in your example:

```
action_client_exe_path_.sendGoal(
        goal,
        boost::bind(&MoveBaseAction::actionExePathDone, this, _1, _2),
        boost::bind(&MoveBaseAction::actionExePathActive, this),
        boost::bind(&MoveBaseAction::actionExePathFeedback, this, _1));
```
But in order to do so, I guess I'll need an instance of the actionclient (in your example this would be `action_client_exe_path_`). How would I get this when using smach_ros.SimpleActionState like I did above? And what are these other three boost::bind-arguments for (what arguments would I have to insert instead)?

corot, 2020-11-02:
If you call exe_path with a smach_ros.SimpleActionState, every time the execute method gets called, it will send a new goal. you just need to take care that the path sent is the latest update.
Or if this is not easily done in your current SM, you can have 2 instances of smach_ros.SimpleActionState, one sending the first path, and the other sending updates.

Ignore the snippet above. It was just illustrating that you can use the the same action client to send goals preempting the previous one: the code is doing what you want: keep updating the current path without aborting navigation.

ML-S, 2020-11-03:
worked, thanks for your help! I'd like to post a full solution but since I changed my approach it doesn't really have anything to do with the question of this issue anymore, therefore I created a [new thread](https://github.com/magazino/move_base_flex/issues/238) which summarizes the solution.

closed 2020-11-03 by ML-S