naturerobots/move_base_flex #232: MBF doesn't preempt when new goal is given

jash101, 2020-10-31:
I'm using rViz to give 2D nav goal, and am using state machine as in the mbf tutorial. It works fine when state is 'WAIT_FOR_GOAL'. But when it is in any other state (GET_PATH, EXE_PATH or RECOVERY) it doesn't preempt the current goal and move on to new goal, as it does in move_base.

corot, 2020-11-02:
Yes, that's expected. The tutorial SM is very simplistic, as it's only intended to serve as an example.
To achieve what you want requires extra work. A couple of ideas on how to do it (don't take it at face value):

- Run a version of WAIT_FOR_GOAL state concurrently, so it can continuously wait for new goals
- [Wrapp your SM with actionlib](http://wiki.ros.org/smach/Tutorials/Wrapping%20a%20SMACH%20Container%20With%20actionlib) and convert RViz goals into calls to the action server

jash101, 2020-11-02:
Thanks! Makes sense. Will try that out 👍

corot, 2020-11-06:
Hope you managed to do it!
Closing this as answered.

closed 2020-11-06 by corot