naturerobots/move_base_flex #240: No plugin loaded with the given name "navfn/NavfnROS"

dean4ta, 2020-11-10:
## Background ##
Hi, I am trying to use move_base_flex to replace move_base so I am using move_base_flex with the `move_base_legacy_relay` node.

## Problem Statement ##
When I publish a `/move_base_simple/goal` message I am running into an error:
`[ERROR] ... The last action goal to "get_path" has been REJECTED`
and a warning:
`[ WARN] ... No plugin loaded with the given name "navfn/NavfnROS"!`

## Initial Debug Notes ##
1. Locating the package `navfn` 
  When I run:
  `rospack find navfn`
  this returns 
  `/opt/ros/melodic/share/navfn`
2. I am able to use move_base fine which also relies on the `navfn` package

## Environment ##
Ubuntu 18
Melodic
Robot is Simulated
Mapping and Localization with slam_toolbox

## Output ##
Here is the entire output from running the move_base_flex node:
```
... logging to /root/.ros/log/ec5c7112-2396-11eb-818a-04ed33dd081a/roslaunch-hopper-1849.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/root/catkin_ws/src/thor_robot_launch/launch/simple_mbf_launch.launch
[1mstarted roslaunch server http://hopper:43105/[0m

SUMMARY
========

CLEAR PARAMETERS
 * /global_costmap/map_server/

PARAMETERS
 * /move_base_flex/GlobalPlanner/allow_unknown: False
 * /move_base_flex/GlobalPlanner/cost_factor: 0.55
 * /move_base_flex/GlobalPlanner/default_tolerance: 0.0
 * /move_base_flex/GlobalPlanner/lethal_cost: 253
 * /move_base_flex/GlobalPlanner/neutral_cost: 66
 * /move_base_flex/GlobalPlanner/old_navfn_behaviour: False
 * /move_base_flex/GlobalPlanner/use_dijkstra: True
 * /move_base_flex/GlobalPlanner/use_grid_path: False
 * /move_base_flex/GlobalPlanner/use_quadratic: False
 * /move_base_flex/GlobalPlanner/visualize_potential: True
 * /move_base_flex/controllers: [{'type': 'eband_...
 * /move_base_flex/dwa/acc_lim_theta: 10.0
 * /move_base_flex/dwa/acc_lim_x: 0.7
 * /move_base_flex/dwa/acc_lim_y: 0.1
 * /move_base_flex/dwa/angular_sim_granularity: 0.025
 * /move_base_flex/dwa/forward_point_distance: 0.0
 * /move_base_flex/dwa/global_frame_id: map
 * /move_base_flex/dwa/goal_distance_bias: 40.0
 * /move_base_flex/dwa/holonomic_robot: False
 * /move_base_flex/dwa/max_scaling_factor: 0.2
 * /move_base_flex/dwa/max_vel_theta: 2.0
 * /move_base_flex/dwa/max_vel_trans: 2.12
 * /move_base_flex/dwa/max_vel_x: 2.0
 * /move_base_flex/dwa/max_vel_y: 0.0
 * /move_base_flex/dwa/min_vel_theta: 0.4
 * /move_base_flex/dwa/min_vel_trans: 0.1
 * /move_base_flex/dwa/min_vel_x: 0.0
 * /move_base_flex/dwa/min_vel_y: 0.0
 * /move_base_flex/dwa/occdist_scale: 0.2
 * /move_base_flex/dwa/oscillation_reset_dist: 0.1
[... 170 lines cut ...]
  <node name="map_server" pkg="map_server" type="map_server" args="$(find thor_robot_launch)/maps/map.yaml" ns="global_costmap" clear_params="true"/> 
  
  <node pkg="mbf_costmap_nav" type="mbf_costmap_nav" respawn="false" name="move_base_flex" output="screen">
    <param name="tf_timeout" value="1.5"/>
    <param name="planner_max_retries" value="3"/>
    <rosparam file="$(find thor_robot_launch)/config/robots/grover/planners.yaml" command="load" />
    <rosparam file="$(find thor_robot_launch)/config/robots/grover/controllers.yaml" command="load" /> 
    <rosparam file="$(find thor_robot_launch)/config/robots/grover/costmap_common_params.yaml" command="load" ns="global_costmap" /> 
    <rosparam file="$(find thor_robot_launch)/config/robots/grover/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find thor_robot_launch)/config/robots/grover/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find thor_robot_launch)/config/robots/grover/global_costmap_params.yaml" command="load" /> 
    <remap from="/cmd_vel" to="/cmd_vel/move_base" />
  </node>
  
  <node name="move_base_legacy_relay" pkg="mbf_costmap_nav" type="move_base_legacy_relay.py"/>

</launch>
```

I have attached my config files if those are helpful
[planners.yaml.txt](https://github.com/magazino/move_base_flex/files/5520232/planners.yaml.txt)
[controllers.yaml.txt](https://github.com/magazino/move_base_flex/files/5520234/controllers.yaml.txt)
[costmap_common_params.yaml.txt](https://github.com/magazino/move_base_flex/files/5520235/costmap_common_params.yaml.txt)
[local_costmap_params.yaml.txt](https://github.com/magazino/move_base_flex/files/5520239/local_costmap_params.yaml.txt)
[global_costmap_params.yaml.txt](https://github.com/magazino/move_base_flex/files/5520242/global_costmap_params.yaml.txt)

dean4ta, 2020-11-10:
#60 
I believe this is a relevant issue

spuetz, 2020-11-10:
The default global planner is 

> base_global_planner (string, default: "navfn/NavfnROS")
But this is not listed in `planners.yaml`
Either add it to the config file or set the `bgp` variable in the `move_base_legacy_relay.py`

closed 2020-11-10 by spuetz

dean4ta, 2020-11-12:
Hi Sebastian,

Firstly thanks for the quick reply yesterday. As you suggested, the solution was to add to the `planners.yaml`:
```yaml
  - name: 'navfn/NavfnROS'
    type: 'navfn/NavfnROS'
```
In addition to that I also added to the `controllers.yaml`:
```yaml
  - name: 'base_local_planner/TrajectoryPlannerROS'
    type: 'base_local_planner/TrajectoryPlannerROS'
```

However, I found a bug with the `move_base_legacy_relay` node with the `bgp`, `blp`, and parameters. When setting up the dynamic reconfigure server [in this line](https://github.com/magazino/move_base_flex/blob/master/mbf_costmap_nav/scripts/move_base_legacy_relay.py#L141) the MoveBaseConfig is used and overwrites all parameters including `bgp`, `blp`. I made a hack fix by only changing `bgp` and `blp` when they are set to `None`, but the rest of my parameters were still overwritten.

I am willing to make changes and submit a PR, but I want to know your thoughts.

Dean

corot, 2020-11-12:
Hi @dean4ta, not sure if I get the latest problem, but,,, what you said is correct, that is, that line triggers a first call to reconfigure that will use all the parameters that you define under 'move_base' namespace. So if you choose a base_local/global_planner, it will be added there:

[image]

Otherwise, the fields will contain the default values:

[image]