naturerobots/move_base_flex #241: ConcurrencySlots for Controllers

dorezyuk, 2020-11-11:
Hey guys,

As far as i understand, we can run multiple controller executions in parallel. This seems really dangerous and I dont know if there is a real use case for that. Is this relevant for any application?

spuetz, 2020-11-11:
Why should this be dangerous? There are applications with multiple robots ;)

dorezyuk, 2020-11-11:
Ah I see. And you would like to control multiple machines from one mbf server? I was thinking that on one machine this would result in something weird. Like having two controllers executing contradicting plans

spuetz, 2020-11-11:
This was the idea, there is still some work to do for this, but currently it not bad having this half read option in. cheers.

closed 2020-11-11 by spuetz