naturerobots/move_base_flex #246: Giving goal for exe_path action in different frame other than map

fayyazpocker, 2020-12-04:
I am publishing directly to _/move_base_flex/exe_path/goal_ as I am only interested in the controller. I am able to give the path with reference to _map_ frame but passing path in a different frame like _base_link_ is not working. I am able to publish the goal without throwing any error but the robot does not follow to the goal unlike when I pass the path in _map_ frame. I can always transform the goal point into map frame and pass it to the path but I think that feature is really good as we have that in move_base.

lucaslins0035, 2022-02-06:
Having the same trouble here! In fact I am trying to pass a goal for the path to be planned in `odom` frame, but it keeps asking for frame `map`. Apparently it does not even support base_link frame with a namespace as `<namespace>/base_link`. It would be nice to have it quickly sorted out just by using the frame names passed into the costmap_2d params.

corot, 2022-11-15:
Sorry, probably I'm too late commenting here, but anyway,

Passing a path referenced to base_link doesn't make sense with the typical controller, as DWA or TEB, as they keep transforming the path into local costmap frame. That means that, as the robot moves, the path moves with it and so it will keep catching the path forever.

That said, MBF just forwards the path to the controller, so she's agnostic about the frame (disclaimer: not 100% sure about this, I need to check the code more thoroughly)

What are the errors you get?  Or no errors, just the robot not moving?

Sorry again for the huge delay in answering