naturerobots/move_base_flex #247: turtlebot3_mbf: No plugin loaded with the given name "navfn/NavfnROS"! and The last action goal to "get_path" has been REJECTED

fjp, 2020-12-06:
Hi, I have troubles to send a 2D nav goal using this package with Gazebo turtlebot3_simulation. When I launch the following launchfiles from [turtlebot3_simulation](https://github.com/ROBOTIS-GIT/turtlebot3_simulations/tree/noetic-devel) and [`turtlebot3_mbf`](https://github.com/Rayman/turtlebot3_mbf) from @Rayman on ROS noetic:

```console
roslaunch turtlebot3_gazebo turtlebot3_world.launch
```

then the `amcl_demo.launch` from [`turtlebot3_mbf`](https://github.com/Rayman/turtlebot3_mbf) package:

```console
roslaunch turtlebot3_mbf amcl_demo.launch
```

And finally setting a 2D Nav Goal in Rviz, turtlebot doesn't move and I get the following error and warning in the terminal:

```console
[ WARN] [1607205100.267987148, 78.750000000]: No plugin loaded with the given name "navfn/NavfnROS"!
[ERROR] [1607205100.268080424, 78.750000000]: The last action goal to "get_path" has been REJECTED
```

<details>
 <summary>Click for the complete terminal output</summary>

```console
roslaunch turtlebot3_mbf amcl_demo.launch
... logging to /home/fjp/.ros/log/ddda414e-3743-11eb-aca7-4d40e2fd6e6c/roslaunch-tensorbook-336152.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://192.168.0.9:37991/

SUMMARY
========

PARAMETERS
 * /amcl/base_frame_id: base_footprint
 * /amcl/gui_publish_rate: 50.0
 * /amcl/initial_pose_a: 0.0
 * /amcl/initial_pose_x: 0.0
 * /amcl/initial_pose_y: 0.0
 * /amcl/kld_err: 0.02
 * /amcl/laser_lambda_short: 0.1
 * /amcl/laser_likelihood_max_dist: 2.0
 * /amcl/laser_max_beams: 180
 * /amcl/laser_max_range: 3.5
 * /amcl/laser_model_type: likelihood_field
 * /amcl/laser_sigma_hit: 0.2
 * /amcl/laser_z_hit: 0.5
 * /amcl/laser_z_max: 0.05
 * /amcl/laser_z_rand: 0.5
 * /amcl/laser_z_short: 0.05
 * /amcl/max_particles: 3000
 * /amcl/min_particles: 500
 * /amcl/odom_alpha1: 0.1
 * /amcl/odom_alpha2: 0.1
 * /amcl/odom_alpha3: 0.1
 * /amcl/odom_alpha4: 0.1
 * /amcl/odom_frame_id: odom
 * /amcl/odom_model_type: diff
 * /amcl/recovery_alpha_fast: 0.0
 * /amcl/recovery_alpha_slow: 0.0
 * /amcl/resample_interval: 1
 * /amcl/transform_tolerance: 0.5
 * /amcl/update_min_a: 0.2
 * /amcl/update_min_d: 0.2
 * /move_base_flex/DWAPlannerROS/acc_lim_theta: 3.2
 * /move_base_flex/DWAPlannerROS/acc_lim_x: 2.5
 * /move_base_flex/DWAPlannerROS/acc_lim_y: 0.0
 * /move_base_flex/DWAPlannerROS/controller_frequency: 10.0
[... 159 lines cut ...]
Found 2 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /gazebo:
   * /gazebo/set_link_state
   * /gazebo/set_model_state
 * /amcl:
   * /initialpose
 * /move_base_legacy_relay:
   * /move_base_simple/goal
   * /move_base/goal
   * /move_base/cancel

WARNING These nodes have died:
 * spawn_urdf-4

Found 1 error(s).

ERROR The following nodes should be connected but aren't:
 * /move_base_flex->/move_base_flex (/move_base_flex/local_costmap/footprint)
 * /move_base_flex->/move_base_flex (/move_base_flex/global_costmap/footprint)
```

Is there anything else that I am missing to get turtlebot3 moving in the simulation?

echo4echo, 2020-12-14:
Hi, @fjp , I met similar issue with you, I didn't check the demo but directly import the move_base_flex into my project. Then run simple goal in rviz:
[image]
Same error, then I took lot time to find the issue and finally got the reason, hopefully help for u:
I simply comment the following code in move_base_legacy_relay.py:
# move_base dynamic reconfigure server
    mb_drs = Server(MoveBaseConfig, mb_reconf_cb) 
Then it works.

I think the reason behind this is:
In: turtlebot3_mbf/launch/amcl_demo.launch, the code load old move base:
 <include file="$(find turtlebot3_mbf)/launch/includes/move_base.launch.xml"/>
In move_base_legacy_relay.py:
it load MoveBaseConfig from move_base.cfg. The move_base.cfg is actually under:
https://github.com/ros-planning/navigation/blob/noetic-devel/move_base/cfg/MoveBase.cfg

If you don't specific anything, it will load the MoveBase.cfg from old move_base lib then of course it couldn't find the default plugin becasue you never define them, (or maybe it got replaced).
You could also check the param  /move_base_legacy_relay/controller_frequence to make sure wether it load the default MoveBase.cfg (20.0) or the customized frequence (5.0).

spuetz, 2020-12-14:
Please see: https://github.com/magazino/move_base_flex/issues/240
and https://github.com/magazino/move_base_flex/issues/60

closed 2020-12-14 by spuetz

spuetz, 2020-12-14:
Btw, at start up of MBF, it says which plugins are loaded.

jtoobias, 2021-07-07:
Hi, the same issue with my own robot and when I try the turtlebot3 tutorial on the MBF wiki. The warning and error I get when I set a goal pose is:
[ WARN] - /move_base_flex: [1625631360.383161930, 216.794000000] No plugin loaded with the given name "navfn/NavfnROS"!
[ERROR] - /move_base_flex: [1625631360.383519016, 216.795000000] The last action goal to "get_path" has been REJECTED

I do have the plugin being accepted:
[ INFO] - /move_base_flex: [1625631349.899322993, 206.333000000] The plugin with the type "navfn/NavfnROS" has been loaded successfully under the name "NavfnROS".
[ INFO] - /move_base_flex: [1625631349.905273725, 206.339000000] Sim period is set to 0.10
[ INFO] - /move_base_flex: [1625631349.942617670, 206.377000000] The plugin with the type "dwa_local_planner/DWAPlannerROS" has been loaded successfully under the name "DWAPlannerROS".

I have attached my turtlebot3 tutorial files:
[turtlebot3_mbf.zip](https://github.com/magazino/move_base_flex/files/6774284/turtlebot3_mbf.zip)
 (which is exactly the same as the wiki one).
 
The only thing that makes it work is by commenting out the line @echo4echo mentioned above, but there must be a proper way.

The system I am using is Ubuntu 18.04, Melodic.
I have looked at other Github answers but either I don't understand them or I feel like I have it already done, but I would think the tutorial would work and provide answers for setting it up. So if anyone can explain the solution clearly I would greatly appreciate it.

Thanks :)

spuetz, 2021-07-07:
Hi @jtoobias. Have a lock at https://uos.github.io/mbf_docs/ 
The reason is that you did not configured it right that MBF or you used the wrong `name` in the action call. At start up \gls{MBF} prints out the names and the types which have been loaded. 
Cheers

spuetz, 2021-07-07:
Hi @jtoobias. Have a lock at https://uos.github.io/mbf_docs/ 
 The reason is that you have not configured MBF correctly or used the wrong name in the action call. At startup MBF prints the names and the types that were loaded.
Cheers

jtoobias, 2021-07-07:
Thank you for the very fast reply @spuetz. I took a look at the mbf_docs link you shared and noticed in all launch files the 
  <node name="move_base_legacy_relay" pkg="mbf_costmap_nav" type="move_base_legacy_relay.py"/> is never run, which is where my problem comes from.
  
I was able to go through the first couple of tutorials as well just to make sure it was working without the node, which I could do. (Also, very nice layout of tutorial on there).

So is this node not required anymore? I thought this node linked the move_base to the new move_base_flex.

Thank you for the help.

denesh-globotix, 2021-10-11:
As mentioned by @echo4echo the problem seems to be coming from move_base_legacy_relay.py. I have added the following parameters under the node's private namespace and the error does not appear any longer. The name to be added is related to the name passed into the mbf_costmap_nav node's yaml file. Examples are show below:
```

planners:
  - name: 'GlobalPlanner'
    type: 'global_planner/GlobalPlanner'

controllers:
  - name: DWAPlannerROS
    type: dwa_local_planner/DWAPlannerROS
```

`  <node name="move_base_legacy_relay" pkg="mbf_costmap_nav" type="move_base_legacy_relay.py">
    <param name="base_global_planner" value="GlobalPlanner" />
    <param name="base_local_planner" value="DWAPlannerROS" />
  </node>`