naturerobots/move_base_flex #253: How to get different outcome codes in GetPath result?

jash101, 2021-01-10:
I like that move_base_flex's GetPath action has extensive outcome codes which can give great additional information on why the path planning failed, which I want to make use of to add case specific recovery. For example, if GetPath outcome is 52 (INVALID START), the robot should back off. If outcome is 53 (INVALID GOAL), then it should plan the path to a nearby pose. 
But after setting up move_base_flex and echoing move_base_flex/get_path/result, while publishing goal to a pose which has an obstacle (the black region), the get path fails with code 55 (PAT_EXCEEDED), although it should be 53 (INVALID GOAL). 

Am I missing something in the setup? Or is it a bug/unfinished feature? I really want to use these outcomes to design state machine to handle each case accordingly using move_base_flex.

Here's my `planners.yaml` file contents:
```
planners:
  - name: 'GlobalPlanner'
    type: 'global_planner/GlobalPlanner'

planner_frequency: 0.5
planner_patience: 5.0

GlobalPlanner:
  orientation_mode: 1
  orientation_window:_size: 1
  publish_potential: true
  lethal_cost: 253
  neutral_cost: 50
  cost_factor: 3.0
```

corot, 2021-01-12:
The problem here is that MBF is just offering the plugins the option of providing detailed outcomes, instead of just success or failure. But it's up to the plugins (planner in this case) to implement the richer interface and provide the error codes.
GlobalPlanner is not currently implementing [MBF interface](https://github.com/magazino/move_base_flex/blob/master/mbf_costmap_core/include/mbf_costmap_core/costmap_planner.h), so it won't provide things like INVALID GOAL, just success or failure.

What you can do is to clone the navigation repo and implement the MBF interface, and return the error codes you consider appropriate.

jash101, 2021-01-13:
Oh, now I get it. Thanks for the reply. Is there a repo/pkg which already does this? Could you help me to understand where I need to make the changes in the planner code?

corot, 2021-03-26:
Hi, sorry for the delay. Indeed, yes!  I modified long time ago global_planner [here](https://github.com/ros-planning/navigation/pull/774)
And [here](https://github.com/magazino/teb_local_planner/commit/e260fd61039dd7a9bc3f86fb3f874e30a4dd34dd)  I added a MBF interface to TEB local planner while maintainng the original nav_core one, so you can now load TEB on both MBF and have the extended interface, but also on old good move base (with the old, very limited interface, only true/false result)

I close this as solved; please reopen if u have further issues

closed 2021-03-26 by corot