naturerobots/move_base_flex #260: Transitioning goal lost cause move_base_flex die

echo4echo, 2021-02-19:
I used the move_base_flex in my project and found a issue in my test. When the robot was sitting the "Black zone" in the global map (That would happen when robot localization is not excatly correct or manually mis-operating), after exceeding the patience time, the move_base_flex would die.
[image]

dorezyuk, 2021-02-19:
Thanks for submitting. Could you upload the coredump from the node?

corot, 2021-04-27:
@echo4echo, can u retry with current master?  I think this issue should be fixed now

echo4echo, 2021-04-27:
@corot 
Hey, Corot, sorry for the late reply. Currently our robot is under maintenance. I will pull down and rerun again when I get the time slot to use the robot.

corot, 2021-07-15:
Hopefully solved with #267; @echo4echo, please reopen if you still find the issue with current master

closed 2021-07-15 by corot