naturerobots/move_base_flex #265: How to load Recovery Behaviors?

jack162, 2021-03-19:
When I set a goal for my real robot. It move a while and stop with some problems.

[ERROR] [1616142355.497759323]: Received plan with zero length
[ WARN] [1616142355.497839757]: Could not transform the global plan to the frame of the controller
[ WARN] [1616142355.498061408]: The controller has been aborted after it exceeded the patience time
[ WARN] [1616142355.498568957]: No Recovery Behaviors loaded!
[ WARN] [1616142355.498666272]: Abort the execution of the controller: Controller failed

It seems like i didn't load the Recovery Behaviors. Could anyone told me how to load it?
thanks a lot.

here is my params files.
[move_base.yaml]
planners:
  - name: 'navfn/NavfnROS'
    type: 'navfn/NavfnROS'

controllers:
  - name: 'base_local_planner/TrajectoryPlannerROS'
    type: 'base_local_planner/TrajectoryPlannerROS'

controller_frequency: 5.0
controller_patience: 3.0

planner_frequency: 1.0
planner_patience: 5.0

oscillation_timeout: 10.0
oscillation_distance: 0.2

spuetz, 2021-03-19:
Try to add this in your `move_base.yaml`
```
recovery_behaviors:
  - name: 'rotate_recovery'
    type: 'rotate_recovery/RotateRecovery'
  - name: 'clear_costmap'
    type: 'clear_costmap_recovery/ClearCostmapRecover
 ```

jack162, 2021-03-19:
**@spuetz Thanks for ur help! I add it and get following ERROR.**

[FATAL] [1616143544.232688643]: Failed to load the clear_costmap_recovery/ClearCostmapRecover recovery behavior, are you sure it's properly registered and that the containing library is built? According to the loaded plugin descriptions the class clear_costmap_recovery/ClearCostmapRecover with base class type mbf_costmap_core::CostmapRecovery does not exist. Declared types are  According to the loaded plugin descriptions the class clear_costmap_recovery/ClearCostmapRecover with base class type nav_core::RecoveryBehavior does not exist. Declared types are  clear_costmap_recovery/ClearCostmapRecovery rotate_recovery/RotateRecovery
[ERROR] [1616143544.232726549]: Could not load the plugin with the name "clear_costmap" and the type "clear_costmap_recovery/ClearCostmapRecover"!

and I try to check if I have clear_costmap_recovery pkg. and it indeeds has.
**/opt/ros/kinetic/share/clear_costmap_recovery**

spuetz, 2021-03-19:
Follow the instructions in the error. Of course you have to install the plugins, e.g. `sudo apt install ros-melodic-clear-costmap-recovery`, and `sudo apt install ros-melodic-rotate-recovery`.

closed 2021-03-19 by spuetz

jack162, 2021-03-19:
I has installed clear-costmap-recovery and still get same error.
(ros-**kinetic**-clear-costmap-recovery is already the newest version)

Anyway thank you very much! i will try to fix it.