naturerobots/move_base_flex #268: Changing frame from which global plan is planned

willie013, 2021-04-28:
Hi there,

I am using move_base_flex with the sbpl_lattice_planner to plan a route for my robot. I want to change the frame from which the route is planned, instead of the standard base_link to another tf named steering.

I thought by changing the `global_costmap/robot_base_frame` parameter to steering the route using sbpl_lattice_planner would get planned from the steering frame, since `global_costmap/robot_base_frame` is used in the getRobotPose of move base flex. However, this is not what seems to be happening even though the robot_base_frame is set correctly.

Using a manual tf transform to change the start pose of makePlan works, but changing the `global_costmap/robot_base_frame` does not work.

Can someone point me in the right direction into what I am missing?

Thanks!

Rayman, 2021-04-28:
You probably also have to change the `robot_frame` parameter, in addition to changing the `*_costmap/robot_base_frame` parameters.

willie013, 2021-04-28:
Thanks Rayman! I tried this and it works flawlessly. However, I could not find the robot_frame parameter in the documentation but only in the source code, is this correct?

closed 2021-04-28 by willie013

Rayman, 2021-04-28:
I'm afraid it's not documented, I just did a quick search for `base_link` in the code :)