naturerobots/move_base_flex #269: Cancel exe_path action is ignored if sent < 100 ms after the goal

corot, 2021-06-02:
To reproduce, set shutdown_costmaps to true, as this introduce a delay between the exe_path goal reception and the execution.

If you send a cancel soon (<100ms) after the goal, we will set the cancel_ flag to true [here](https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/src/abstract_controller_execution.cpp#L263), but then the execution start reset it to false [here](https://github.com/magazino/move_base_flex/blob/master/mbf_abstract_nav/src/abstract_execution_base.cpp#L67).

I tried to remove these 3 lines under "reset the member vars" and now cancel works regardless of how fast u send it after the goal. I think these lines are not needed, as we create a new execution object each time. @spuetz, can u confirm?

spuetz, 2021-06-10:
As the controller-execution object will be created for a start event, we can safely remove the` cancel_ = false;` in line 67.

corot, 2021-07-15:
Fixed with #270

corot, 2021-07-15:
Fixed with #270

closed 2021-07-15 by corot