naturerobots/move_base_flex #277: move_base_legacy_relay.py cancel navigation implementation

fllay, 2021-09-06:
It seems that move_base_legacy_relay.py  does not relay  /move_base/cancel to /move_base_flex/move_base/cancel. Or did I miss something here?

Rayman, 2021-09-08:
You are probably right. As far as I can see, the relay doesn't handle preemption. That would be a good candidate for a PR.

ju-mingyue, 2021-11-22:
@Rayman  When I use the move_base_flex function package again, I follow the link [https://github.com/uos/ceres_robot/blob/master/ceres_navigation/launch/move_base_flex.launch](url) to configure the file, but after running it, it is found in the node graph that this function package does not subscribe to the map and the lidar signal (the topic here is corresponding), and I can’t read it To the planned path.I don’t quite understand what does move_base_legacy_relay do?

Rayman, 2021-11-22:
The interface of `move_base_flex` is `get_path` and `exe_path`. This is different from the old `move_base`. So there is a node `move_base_legacy_relay` that makes `move_base_flex` also provide the old interface for backwards compatibility.

With this you have a drop-in replacement for `move_base`. But when you start using `move_base_flex` fully, you won't need it anymore.

ju-mingyue, 2021-11-22:
@Rayman 
Thank you very much for your prompt reply. Now my function package causes the robot to not run continuously. After the target point is reached, the robot will not run if the command is issued again.
[image]
Below is the running file of my node graph and configuration：
[image]
[image]

Look forward to your reply。

closed 2021-12-02 by spuetz