naturerobots/move_base_flex #282: Which C++ language version?

mhepeyiler, 2021-11-01:
Hi, I want to contribute to the project. However I could not find anything about the C++ version, I have checked CMake files. Which one do you use? Also, do you have any benchmark tools? Thanks in advance.

Rayman, 2021-11-02:
(correct me if I'm wrong) I think we support melodic so the code should be C++11 according to [REP 3](https://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021). Right now I couldn't find any C++11 usages but that shouldn't be a problem.

mhepeyiler, 2021-11-02:
Thanks for your answer. I think you are wrong. According to REP3, melodic targets to C++14. I will consider the language is C++14.

Rayman, 2021-11-04:
You are right, it seems to be C++14

closed 2021-11-05 by mhepeyiler