naturerobots/move_base_flex #290: move_base action: cancel previous goal if get_path fails upon receiving a new one

corot, 2022-01-23:
Current behavior:

1. send a valid goal with RViz
2. while the robot goes there, send another goal to an invalid place, e.g. a wall
3. get_path will fail, executing the recovery behaviors
4. but robot will keep going to the original goal

Not a big issue, but current is not the expected behavior:

1. send a valid goal with RViz
2. while the robot goes there, send another goal to an invalid place, e.g. a wall
3. get_path will fail, executing the recovery behaviors
4. robot stops and fail with a GetPathResult error

corot, 2022-01-23:
I have pushed the fix on master by mistake :see_no_evil:, sorry
https://github.com/magazino/move_base_flex/commit/a5c19cb8f3135679e4f8f36a9ec6123b121d69ab
I don't think it's worth doing a push force to revert it, so @dorezyuk , @spuetz , can u make a postmortem review, just in case I need to revert it?  Thanks!

spuetz, 2022-01-23:
> I have pushed the fix on master by mistake see_no_evil, sorry [a5c19cb](https://github.com/magazino/move_base_flex/commit/a5c19cb8f3135679e4f8f36a9ec6123b121d69ab) I don't think it's worth doing a push force to revert it, so @dorezyuk , @spuetz , can u make a postmortem review, just in case I need to revert it? Thanks!

looks good to me. -- Sebastian

dorezyuk, 2022-01-25:
also good for me

corot, 2022-01-25:
:relieved:

corot, 2022-01-25:
Fixed with a5c19cb

closed 2022-01-25 by corot