naturerobots/move_base_flex #303: ExePath Action does not follow the path

minghaohsu410168, 2022-08-17:
Hi everyone, 

I have used the ExePath Action to publish the stored path for navigation, but the robot didn't follow the predefined path and instead directly navigates to the final points.

For example, the path I published contains four points {(0, 0), (1, 0), (2, 0), (3, 0)}. When the initial pose of robot is (1, 1), after getting the action goal, the robot navigates to (3, 0) instead of navigates to (0, 0) and then go to (1, 0), (2, 0), (3, 0)

Below is part of my code, please give me some advise thank you.
```Python

    ## move_base_flex exe path client
    client = actionlib.SimpleActionClient('/move_base_flex/exe_path',ExePathAction)
    rospy.loginfo("Connected to Move Base Flex ExePath server!")

    client.wait_for_server()
    goal = ExePathGoal()
    path_ = Path()

    poses_ = []
    p = PoseStamped()
    p.header.frame_id = "map";
    p.pose.position.x = 0.0;
    p.pose.position.y = 0.0;
    p.pose.orientation.z = 0; 
    p.pose.orientation.w = 1;    
    p.header.stamp = rospy.Time.now()
    poses_.append(p)

    p = PoseStamped()
    p.header.frame_id = "map";
    p.pose.position.x = 1.0;
    p.pose.position.y = 0.0;
    p.pose.orientation.z = 0; 
    p.pose.orientation.w = 1;    
    p.header.stamp = rospy.Time.now()
    poses_.append(p)

    p = PoseStamped()
    p.header.frame_id = "map";
    p.pose.position.x = 2.0;
    p.pose.position.y = 0.0;
    p.pose.orientation.z = 0; 
    p.pose.orientation.w = 1;    
    p.header.stamp = rospy.Time.now()
    poses_.append(p)

    p = PoseStamped()
    p.header.frame_id = "map";
    p.pose.position.x = 3.00;
    p.pose.position.y = 0.0;
    p.pose.orientation.z = 0; 
    p.pose.orientation.w = 1;    
    p.header.stamp = rospy.Time.now()
    poses_.append(p)

    path_.header.frame_id = "map";
    path_.poses = poses_;
    path_.header.stamp = rospy.Time.now();
    path_pub.publish(path_)
```

Rayman, 2022-08-22:
The behavior of the robot depends on which local planner you're using. DWA for example will cut corners. If you want this behavior you have to probably use a different local planner.

minghaohsu410168, 2022-08-23:
Thank you for your reply, I am currently using teb_local_planner. But the followed path doesn't match the global plan, I think the reason is still local planner. Should I keep adjusting the parameters or directly try another local planner?

corot, 2022-11-15:
Most controllers (including TEB) expect a dense path, with poses close to each other (normally <= 10 cm)

Your path is very sparse, so far from what TEB expects.

You can for example create a smooth path that links your 4 points and pass it to TEB, maybe using something like [this](https://github.com/gkouros/path-smoothing-ros)  (disclaimer: just found it doing a fast search for "path smoothing spline")

That said, this is not a MBF issue, but a TEB limitation, so I'll close this (but feel free to ask or comment!)
ROS Answers is a more appropriate forum

closed 2022-11-15 by corot

PaddyCube, 2023-08-27:
I have the same issue and don't understand what path dense makes a difference.
In my opinion, ExePath should send each pose as single goal so global and local planner can work with. If So, how can the local planner know the entire path and shortcut it?