naturerobots/move_base_flex #328: Failed to load wave front planner and mmp planner

JACKLiuDay, 2023-08-31:
Hi, thank you for your great job about mbf. My PC is ubuntu 20.04 with noetic. I install mbf by sudo apt install ros-noetic-move-base-flex and mesh-navigation, pluto robot... But when I launch the pluto robot simulation, something wrong happened. 
 roslaunch pluto_gazebo pluto_botanical_garden.launch
roslaunch pluto_navigation botanical_garden_osnabrueck.launch
[image]

12ahmed52, 2024-04-24:
hello did you solve this issue ? i have the same issue

Rayman, 2024-04-24:
According to your error message, the planner is not found. This means either:
- The planner is not installed or compiled
- The planner doesn't export the plugin correctly

Maybe you could file your bug into the bugtracker of the planner?

closed 2024-04-26 by JACKLiuDay