naturerobots/move_base_flex #346: Finalize ROS2 Migration

Cakem1x, 2024-10-16:
The [humble branch](https://github.com/naturerobots/move_base_flex/tree/humble) currently tracks the latest move base flex version for ROS2.

I want to tackle the following post migration ToDos:
* update default branch: from `master` to `humble`
* update readmes (ROS2 readme (`humble`), ROS1 readme (`master`))
  * add logos for companies
  * add/update notice about and link to ROS1 and ROS2 branch

Cakem1x, 2024-10-16:
Before changing the default branch to `humble`, I'd like to get confirmation from other maintainers on whether that will break anything critical (@spuetz @corot ).
The rationale for this change is that it makes the landing of new people on the [repo's main page](https://github.com/naturerobots/move_base_flex) nicer, assuming most of them are interested in using ROS2.

corot, 2024-10-18:
> Before changing the default branch to `humble`, I'd like to get confirmation from other maintainers on whether that will break anything critical (@spuetz @corot ). The rationale for this change is [...]

I think the default branch should always be `master`. But if most new developments will be on ROS 2, it makes sense to 
- rename master as noetic
- rename humble as master
But as migration is still wip (costmap server missing), we need to wait to do that

Besides that, I think there are changes in jazzy that break compatibility with costmaps, so future master should work for jazzy, not humble

amock, 2025-04-12:
@corot, @Cakem1x

As we've got a few side projects going on that use different map representations (like mesh maps, OpenVDB), it got me thinking: do we really need to include a costmap_2d implementation in this repo?

The more I thought about it, the more I liked the idea of **not** providing any specific map representation in this repo at all.
With solid docs, I actually think it makes it easier to understand what MBF is really about. It's not limited to a particular map representation.

Otherwise, people might confuse MBF with another ROS 2 framework for navigation on 2D costmaps.
One thing that actually gets hard is testing. However, I really think we can find a good solution to that.

What do you think of that idea?

Cakem1x, 2025-04-14:
I am good with keeping implementations for different concrete map representations in **separate repos**. Especially if they come with a lot of additional dependencies. Having an overview of where to find map specific implementations would be very important, though.

I currently do not plan to port the **costmap2d** server, as we are not using it at the moment. So finalizing the ROS2 version would be good, as it is very much usable already (we are running the simple nav server and the mesh nav server). When we want to plan on 2d gridmaps, we will likely end up using another gridmap implementation.

Regarding **testing**, for ROS2, I added a bit of code that "simulates" a robot in a very crude way, see [mbf_test_utility](https://github.com/naturerobots/move_base_flex/tree/humble/mbf_test_utility). Using this, I added a [simple_nav_integration_test](https://github.com/naturerobots/move_base_flex/blob/humble/mbf_simple_nav/test/simple_nav_integration_test.cpp). I think simple_nav should remain part of the repo, at least to have something testable that instantiates the interface.
Implementations for other map representations could use this test as a blueprint for an integration test.